JokerJohn/LIO_SAM_6AXIS #3: How does the system initialize without the gravity direction indicated by the 9-axis IMU

jiejie567, 2022-05-09:
At first, thank you for publishing the great work, which is of great benefit to the poor people who only have   6-axis IMU like me. 
But I have a silly question, the lio-sam get the gravity direction by the  9-axis IMU when initializing. Thus if I apply your code in my project, will I need to keep my IMU horizontal？

JokerJohn, 2022-05-09:
Of course, while you're collecting data, we strongly recommend that you keep the IMU horizontally stationary for a few seconds at the beginning.In addition, If you are using GPS, you also need to start collecting data when the GPS LLA covariance is small.

JokerJohn, 2022-05-09:
> At first, thank you for publishing the great work, which is of great benefit to the poor people who only have 6-axis IMU like me. But I have a silly question, the lio-sam get the gravity direction [...]

Of course, while you're collecting data, we strongly recommend that you keep the IMU horizontally stationary for a few seconds at the beginning.In addition, If you are using GPS, you also need to start collecting data when the GPS LLA covariance is small.

jiejie567, 2022-05-09:
@JokerJohn thank you for your quick and accurate answer！

closed 2022-05-09 by jiejie567