JokerJohn/LIO_SAM_6AXIS #4: 我最近也要使用6轴IMU和雷达进行试验，请问用你的这种方法需要对IMU和雷达进行标定吗 43/5000  I will also use 6-axis IMU and radar for test recently, may I ask if I need to calibrate IMU and radar with your method

wangxb-code, 2022-05-11:


JokerJohn, 2022-05-11:
> 

Of course，you first need to calibrate the extrinsic between IMU and Lidar. Then you also need to calibrate the intrinsics of IMU，that is analyze the random walk noise and bias of IMU.

wangxb-code, 2022-05-11:
谢谢你的回答，我是这方面的新手，我想请问一下你的IMU数据是怎么获得的，IMU的数据格式需要是sensor::imu数据格式吗，标定IMU的时候需要的数据类型是sensor::imu数据类型吗，因为我通过CAN通讯获得了IMU的Motorola的数据类型，请问如何将它转换为sensor::imu数据类型
Thank you for your answer, I am a novice in this field, I would like to know how you get your IMU data, should the data format of IMU be sensor:: msg data format? whether the data type required for calibrating IMU is sensor:: IMU data type. Because I obtained the data type of IMU from Motorola through CAN communication, may I ask how to convert it to sensor:: IMU data type？

JokerJohn, 2022-05-11:
1.write a ros driver to parse the imu raw data from Can port and publish these messages.
2.strictly sync your Imu and lidar sensor. Make sure the imu timestamp is correct and there are no missing frames.

closed 2022-05-11 by JokerJohn