JokerJohn/LIO_SAM_6AXIS #8: RTK and  lio has yaw offset.  roslaunch lio_sam_6axis run.launch  with hkust_20201105full.bag.

lucianzhong, 2022-06-09:


lucianzhong, 2022-06-09:
using  run.launch with hkust_20201105full.bag, and there is an yaw offset between rtk path and lio path. Please have a look. Thanks a lot.

[image]

JokerJohn, 2022-06-09:
@lucianzhong  fact, this has no effect, you ONLY need to wait a while for it to automatically align. Because I adopted a new strategy in the GPS initialization module, and optimized it together after accumulating 20 GPS factors of good quality. This can avoid many problems, since our GPS data is not stable encough, and it is not RTK-GPS.

lucianzhong, 2022-06-09:
Thanks a lot for the quick reply.  
我这边跑您提供的bag，刚开始lio很稳定，但是和gps的yaw偏差很大，然后跑一段时间后，到了一个空旷的地方，lio会乱跳，完全不能定位了，和gps的path不能align

lucianzhong, 2022-06-09:
想请问一下，您那边有没有遇到过类似的问题，或者一些代码的提示，非常感谢。

JokerJohn, 2022-06-09:
> Thanks a lot for the quick reply. 我这边跑您提供的bag，刚开始lio很稳定，但是和gps的yaw偏差很大，然后跑一段时间后，到了一个空旷的地方，lio会乱跳，完全不能定位了，和gps的path不能align

确认你用的是最新版本的code，如果还是这个问题的话，我再检查一下代码版本。

JokerJohn, 2022-06-09:
> 想请问一下，您那边有没有遇到过类似的问题，或者一些代码的提示，非常感谢。

我刚更新了代码，确认无误，详细的视频教程将在2个小时后完成上传。
[image]

JokerJohn, 2022-06-09:
> 想请问一下，您那边有没有遇到过类似的问题，或者一些代码的提示，非常感谢。

按这个视频操作： https://github.com/JokerJohn/LIO_SAM_6AXIS/issues/8#issuecomment-1150696204

lucianzhong, 2022-06-09:
非常感谢，我试一下

lucianzhong, 2022-06-09:
可以了，太感谢了，  star, fork ,follow 一键三连了  :smile:

JokerJohn, 2022-06-09:
> 可以了，太感谢了， star, fork ,follow 一键三连了 smile

要适配自己的传感器数据的话，我在readme中也更新了操作视频链接！（B站稍晚，需要审核）

closed 2022-06-09 by JokerJohn

lucianzhong, 2022-06-09:
期待  :stuck_out_tongue_winking_eye: