JokerJohn/LIO_SAM_6AXIS #10: gps_odom与后端优化的轨迹不重合

weixiaopassking, 2022-06-21:
你好，非常感谢在方向初始化方面提供的思路帮助，但是还有一个问题想请教，如下图如果在gps信号良好的时候，我们认为gps_odom轨迹为真值，但是后端优化的轨迹与gps_odom却不重合，这是什么原因呢？理论上如果lidar_odom与gps_odom已经对齐过，他们俩的轨迹应该有较高重合才对，期待解答，谢谢。
[image]

JokerJohn, 2022-06-21:
> 你好，非常感谢在方向初始化方面提供的思路帮助，但是还有一个问题想请教，如下图如果在gps信号良好的时候，我们认为gps_odom轨迹为真值，但是后端优化的轨迹与gps_odom却不重合，这是什么原因呢？理论上如果lidar_odom与gps_odom已经对齐过，他们俩的轨迹应该有较高重合才对，期待解答，谢谢。 [image]

如果是RTK状态，GPS确实可以作为真值，但在户外场景下要保持GPS RTK持续FIX状态，不太现实，会受到诸多干扰，所以无法确保每一帧GPS位置数据都是准确的。而SLAM局部位姿的估计上比较精准，相当于可以用SLAM来补足这些质量较低的GPS点。
至于为什么我给的数据中GPS ODOM轨迹没有和SLAM融合后的轨迹重合，有以下原因。
1.数据是普通的GPS，无RTK，大部分场景GPS质量较差，GPS_ODOM只是粗略的估计了yaw朝向，没有得到每一个位置的精准位姿。
2.GPS数据不稳定，z轴相比XY误差更大，即使在同一个地点也会因为不同时间到达得到有一定差异的数据。在突然从信号好切到信号不好的位置，需要一定时间的收敛。所以即使是RTK FIX状态，短时间的GPS数据也不一定能用。所以在做优化的时候需要慎重选取GPS点，在GPS信号一般的局部位置上我们更相信SLAM的结果。
3.lidar_odom与GPS odom对齐并不意味着两者轨迹重合。和你选取用于优化的GPS点的质量、数量以及权重有关系。

JokerJohn, 2022-06-21:
如果你想在信号好的时候，GPS估计与SLAM轨迹完全重合，可以再这些位置吧GPS权重设置大一点，对应修改gtsam中noisemodel。

closed 2022-06-23 by JokerJohn