JokerJohn/LIO_SAM_6AXIS #13: Some topic is lost in hkust_20201105full.bag

chengwei0427, 2022-07-26:
Hi, @JokerJohn 
Thank you for your great work and the test dataset.
Some data is lost in the last 100 seconds of hkust_20201105full.bag. This may cause a error.
[image]

chengwei0427, 2022-07-26:
> > Hi, @JokerJohn Thank you for your great work and the test dataset. Some data is lost in the last 100 seconds of hkust_20201105full.bag. This may cause a error. [image]
> 
> Yes, although about 30 frames of point clouds are lost in the last part, it has little effect on the performance of the algorithm. I will release a test bag of ouster128+stim300 soon.

Glad to hear that.

JokerJohn, 2022-07-26:
> Glad to hear that.

new dataset has released, you can download it here.
[image]

chengwei0427, 2022-09-06:
你好， @JokerJohn 

1. 请问你分享的如下两个bag，激光的时间戳，是激光帧头还是帧尾的时间呢？
```
  hkust_20201105full.bag
  20220216_garden_day_ref_compressed.bag
```
2. 20220216_garden_day_ref_compressed.bag 好像没有gps数据？ 请问你有计划开源一个带gps和轮速计的数据集吗？

JokerJohn, 2022-09-06:
> 你好， @JokerJohn 
> 
> 
> 
> 1. 请问你分享的如下两个bag，激光的时间戳，是激光帧头还是帧尾的时间呢？
> 
> ```
> 
>   hkust_20201105full.bag
> 
>   20220216_garden_day_ref_compressed.bag
> 
> ```
> 
> 2. 20220216_garden_day_ref_compressed.bag 好像没有gps数据？ 请问你有计划开源一个带gps和轮速计的数据集吗？

1.hkust数据是velodyne 16，其雷达点云中每个点的时间戳我印象中应该是（真实时间戳-帧头时间戳），数值应该小于0.1s。garden day数据是ouster雷达，其时间戳是处理后的，time通道中存储的时间戳也应该是(真实时间戳-帧头)，数值小于0.1s
2.garden day数据集基本在室内手持采集，没有gps数据，也没有轮速。其他带🈶️轮速的车载平台数据后续将在新的数据集中发布出来。

chengwei0427, 2022-09-06:
Thank you very much! Question solved. I'll close this issue.

closed 2022-09-06 by chengwei0427