JokerJohn/LIO_SAM_6AXIS #14: error: ‘make_unique’ is not a member of ‘std’

kayousolar, 2022-07-27:
Hello, thank you for sharing your work. I encountered this error when compiling. May I ask for you some advice?

My operating system is Ubuntu18.04
The C compiler identification is GNU 11.1.0
The CXX compiler identification is GNU 11.1.0

/home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:29: error: ‘make_unique’ is not a member of ‘std’
  256 |         dataSaverPtr = std::make_unique<DataSaver>(saveDirectory, sequence);
      |                             ^~~~~~~~~~~
/home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:29: note: ‘std::make_unique’ is only available from C++14 onwards
/home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:50: error: expected primary-expression before ‘>’ token
  256 |         dataSaverPtr = std::make_unique<DataSaver>(saveDirectory, sequence);

JokerJohn, 2022-07-27:
> Hello, thank you for sharing your work. I encountered this error when compiling. May I ask for you some advice?
> 
> My operating system is Ubuntu18.04 The C compiler identification is GNU 11.1.0 The CXX compiler identification is GNU 11.1.0
> 
> /home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:29: error: ‘make_unique’ is not a member of ‘std’ 256 | dataSaverPtr = std::make_unique(saveDirectory, sequence); | ^~~~~~~~~~~ /home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:29: note: ‘std::make_unique’ is only available from C++14 onwards /home/xwcd/Project/LIO_SAM_6AXIS-main/src/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:256:50: error: expected primary-expression before ‘>’ token 256 | dataSaverPtr = std::make_unique(saveDirectory, sequence);

make_unique is an C++14 feature. Using the following directive in CMakeLists.txt fixed it for you:
`set(CMAKE_CXX_STANDARD 14).`

closed 2022-07-27 by JokerJohn

Kin-Zhang, 2022-07-29:
fix on 48e2dbf323d7f285631306531ba04307414e58bb