JokerJohn/LIO_SAM_6AXIS #29: Using bag file created from save_map service.

Dhruv0208, 2022-10-17:
Can you guide how to visualize the bag file generated. The bag file generated show to tf data in it, due to which I am not able to visualize in the rviz.

Dhruv0208, 2022-10-18:
Also, when calling the save map service, the global map with respect to lidar as well as imu, is not being saved, and they are also missing in the screenshot you have posted as an example.

JokerJohn, 2022-10-29:
> Also, when calling the save map service, the global map with respect to lidar as well as imu, is not being saved, and they are also missing in the screenshot you have posted as an example.

use the newer version codes, 
[image]

closed 2022-10-29 by JokerJohn

Dhruv0208, 2022-11-14:
Hi,
> Also, when calling the save map service, the global map with respect to lidar as well as imu, is not being saved, and they are also missing in the screenshot you have posted as an example.

The above issue, is still persistent can you help with it? 
  bool saveMapService(lio_sam_6axis::save_mapRequest& req, lio_sam_6axis::save_mapResponse& res)  The mentioned function in the mapOptimization.cpp file is not running.