JokerJohn/LIO_SAM_6AXIS #31: Runing LIO_SAM_6AXIS on other dataset

MoonSheepJx, 2022-11-02:
您好，首先非常感谢您这么棒的工作
我想使用LIO_SAM_6AXIS在其它数据集上进行测试，并且开启GPS
1、但是我在运行时，发现了GPS与系统估计的状态差别较大，请问这是正常现象嘛，还是我的一些参数设置错误的问题？
[image]
2、我看了一下源代码和您在其他issue下的回答，貌似开启`useGPS`，一定是要等待GPS信号，才能初始化整个系统嘛，那如下这个现象就是正常的吧
[image]

（我的配置文件如下，数据集为M2DGR）
[image]
再次非常感谢您的工作，非常希望您能给予一些帮助！

MoonSheepJx, 2022-11-02:
您好，非常感谢您的及时回复与帮助，请问我可以这样理解吗
只有当gps拥有足够的帧数，初始化后，系统轨迹才受到gps的约束，此时的系统输出轨迹，才是结合了gps约束的轨迹，而当gps未初始化前，系统轨迹相当于纯LIO的轨迹

JokerJohn, 2022-11-02:
> 您好，非常感谢您的及时回复与帮助，请问我可以这样理解吗 只有当gps拥有足够的帧数，初始化后，系统轨迹才受到gps的约束，此时的系统输出轨迹，才是结合了gps约束的轨迹，而当gps未初始化前，系统轨迹相当于纯LIO的轨迹

是的

MoonSheepJx, 2022-11-02:
> > 您好，非常感谢您的及时回复与帮助，请问我可以这样理解吗 只有当gps拥有足够的帧数，初始化后，系统轨迹才受到gps的约束，此时的系统输出轨迹，才是结合了gps约束的轨迹，而当gps未初始化前，系统轨迹相当于纯LIO的轨迹
> 
> 是的

非常感谢您的帮助！

closed 2022-11-03 by JokerJohn