JokerJohn/LIO_SAM_6AXIS #33: save traj on my own dataset

MoonSheepJx, 2022-11-03:
您好！
我根据您的readme在运行算法后保存了轨迹
有几点疑问相您请教
[image]

1、这个`odo_tum.txt`和`optimized_odom_tum.txt`有什么区别嘛，我用evo查看他们是没有任何区别的。
2、另外一个问题比较困扰我，昨天询问了您GPS初始化后主导轨迹的问题
我也调整了gps参数使其能初始化成功，我发现在gps初始化前，gps轨迹和LIO轨迹是有一些差距的，在初始化后会有一个轨迹跳变，LIO的轨迹向gps轨迹靠近，那进行最后的轨迹保存，是跳变之后，带有gps约束的轨迹吗？( 我是在数据集跑完的情况下执行的rosservice call )
3、我用evo查看了保存的轨迹的pose个数，相比其他算法或地面真值都是较少的，想向您询问一下这种轨迹pose个数少的情况是不是正常的
如下图终端的底部所示
[image]
但是整体轨迹是一致的（如下图）
[image]
最后，非常感谢您的工作，希望得到您的建议。

JokerJohn, 2022-11-03:
> 您好！ 我根据您的readme在运行算法后保存了轨迹 有几点疑问想向您咨询一下 [image]
> 
> 1、这个`odo_tum.txt`和`optimized_odom_tum.txt`有什么区别嘛，我用evo查看他们是没有任何区别的。 2、另外一个问题比较困扰我，昨天询问了您GPS初始化后主导轨迹的问题 我也调整了gps参数使其能初始化成功，我发现在gps初始化前，gps轨迹和LIO轨迹是有一些差距的，在初始化后会有一个轨迹跳变，LIO的轨迹向gps轨迹靠近，那进行最后的轨迹保存，是跳变之后，带有gps约束的轨迹吗？( 我是在数据集跑完的情况下执行的rosservice call ) 3、我用evo查看了保存的轨迹的pose个数，相比其他算法或地面真值都是较少的，想向您询问一下这种轨迹pose个数少的情况是不是正常的 如下图终端的底部所示 [image] 但是整体轨迹是一致的（如下图） [image] 最后，非常感谢您的工作，希望得到您的建议。

1.前者只是odomerty，后者是pgo优化之后的。最终保存的是施加了gps约束后的最终轨迹（天蓝色轨迹） 。
2.Lio-sam是实时性较好的系统，在进行后端优化的时候通过距离和角度变化选取了关键帧，保证factor graph规模不至于太大。如果你想位姿稠密一些，可以调整下面两个参数。
[image]

MoonSheepJx, 2022-11-03:
明白了！非常感谢您的回复!

closed 2022-11-03 by JokerJohn