JokerJohn/LIO_SAM_6AXIS #38: password of GT

coolaogege, 2022-12-02:
Thank you for your nice work! 
I have tested this code , it works perfectly ！Now i want to evaluate this trajectory by evo. But i can not find this password of GT. Can you tell me the password？By the way,  How are the ground truth values obtained for these two datasets？
[image]

JokerJohn, 2022-12-03:
[link](https://hkustconnect-my.sharepoint.com/:t:/g/personal/xhubd_connect_ust_hk/ESoJj5STkVlFrOZruvEKg0gBasZimTC2HSQ2kqdIOWHiGg?e=Hup1fT)

you can try the password 123

JokerJohn, 2022-12-03:
[image]

coolaogege, 2022-12-04:
> [link](https://hkustconnect-my.sharepoint.com/:t:/g/personal/xhubd_connect_ust_hk/ESoJj5STkVlFrOZruvEKg0gBasZimTC2HSQ2kqdIOWHiGg?e=Hup1fT)
> 
> you can try the password 123

Thanks, I can get GT!

closed 2022-12-04 by coolaogege