JokerJohn/LIO_SAM_6AXIS #40: gps因子图

chenjianqiang199526, 2022-12-19:
您好，请问下为啥累计20个gps因子加入的时候就会崩掉？是前面累计的时候pose不能有太大的变化？
[image]

JokerJohn, 2022-12-19:
> 您好，请问下为啥累计20个gps因子加入的时候就会崩掉？是前面累计的时候pose不能有太大的变化？ [image]

LIO与GNSS的初始化需要足够的translation.

chenjianqiang199526, 2022-12-20:
> > 您好，请问下为啥累计20个gps因子加入的时候就会崩掉？是前面累计的时候pose不能有太大的变化？ [image]
> 
> LIO与GNSS的初始化需要足够的translation.

不太明白您的意思，是对速度有要求吗？我的实验模型车在直线上走了一段时间到了累计20次的时候也会直接崩掉

JokerJohn, 2022-12-20:
> > > 您好，请问下为啥累计20个gps因子加入的时候就会崩掉？是前面累计的时候pose不能有太大的变化？ [image]
> > 
> > 
> > LIO与GNSS的初始化需要足够的translation.
> 
> 不太明白您的意思，是对速度有要求吗？我的实验模型车在直线上走了一段时间到了累计20次的时候也会直接崩掉

初始化模块本质上是对齐LIO轨迹和GPS轨迹，所以在初始阶段，尽可能得等车辆多运动一段距离，有了足够的GNSS观测，一块来优化。根据我的经验来看，崩掉的原因不是Gnss观测不够，而是你的Gnss协方差阈值设置得不太对。你可以先观察Gps topic的输出，确定在gps的协方差在什么大小的时候，gps定位结果比较可靠，然后修改yaml中的这个阈值即可。

chenjianqiang199526, 2022-12-21:
> 

我的协方差设置的是0，认为全是可信的，有没可能是坐标系的问题，我看gps的position是按东北天坐标系，估计的pose好像是雷达的坐标系

JokerJohn, 2022-12-21:
> > 
> 
> 
> 
> 我的协方差设置的是0，认为全是可信的，有没可能是坐标系的问题，我看gps的position是按东北天坐标系，估计的pose好像是雷达的坐标系

肯定不能全信，gps数据本身就不稳定，任何一个不稳定的gps factor，（比如局部出现频繁跳动）在优化中都可能导致gtsam崩溃。lio坐标系可以理解成局部随机开始的，起始位姿为单位阵，但起点就是local系的坐标原点，gps转换到local系下与lio有一个yaw-offset，初始化就是完成这个的。以我的经验看，不是坐标系的问题。

chenjianqiang199526, 2022-12-21:
> > > 
> > 
> > 
> > 我的协方差设置的是0，认为全是可信的，有没可能是坐标系的问题，我看gps的position是按东北天坐标系，估计的pose好像是雷达的坐标系
> 
> 肯定不能全信，gps数据本身就不稳定，任何一个不稳定的gps factor，（比如局部出现频繁跳动）在优化中都可能导致gtsam崩溃。lio坐标系可以理解成局部随机开始的，起始位姿为单位阵，但起点就是local系的坐标原点，gps转换到local系下与lio有一个yaw-offset，初始化就是完成这个的。以我的经验看，不是坐标系的问题。

确实是转角处gps不稳定造成的，多谢大佬的指导

closed 2022-12-21 by chenjianqiang199526