JokerJohn/LIO_SAM_6AXIS #41: 单帧点云保存

ooooohhhhhxxx, 2023-01-04:
您好，很棒的项目！

有一些关于地图保存疑问，因为我在做完slam后，还需要对单帧进行相应的处理，所以在使用原版lio-sam，我通过修改代码，将所有帧都加入key frame并保存，但是保存出来的帧跟原始数据中的帧不能一一对应，例如我使用kitti raw数据的2011_09_26_0104中，一共有312帧，通过上述操作，并将rosbag play设置参数 -r 0.1降低播放速度，最终保存的帧数只有310帧。请问在您的项目中，能否有方法将bag包中的所有帧做完slam后，全部保存吗？

JokerJohn, 2023-01-04:
> 

这部分应该是同步锁的问题，我还没来得及更新。这两天我更新一下。

ooooohhhhhxxx, 2023-01-04:
> 这部分应该是同步锁的问题，我还没来得及更新。这两天我更新一下。

好的，非常感谢

JokerJohn, 2023-01-04:
> 您好，很棒的项目！
> 
> 有一些关于地图保存疑问，因为我在做完slam后，还需要对单帧进行相应的处理，所以在使用原版lio-sam时，通过修改代码，将所有帧都加入key frame并保存，但是保存出来的帧跟原始数据中的帧不能一一对应，例如我使用kitti raw数据的2011_09_26_0104中，一共有312帧，通过上述操作，并将rosbag play设置参数 -r 0.1降低播放速度，最终保存的帧数只有310帧。请问在您的项目中，能否有方法将bag包中的所有帧做完slam后，全部保存吗？

保存地图的时候会保存相应的result.bag结果，你可以在这个result.bag中提取去除畸变的点云，它的数量应该和关键帧pose一致。

closed 2023-03-18 by JokerJohn