JokerJohn/LIO_SAM_6AXIS #44: 基于地图定位

luameows, 2023-03-15:
感谢大佬的工作。
想请教下，通过gps融合建图后，基于地图的ndt输出的位置是在哪个坐标系呢，似乎和enu坐标系还有些偏移？这导致在定位时候融合gps会出现坐标系偏差，有什么解决方法么？

JokerJohn, 2023-03-16:
构建的地图在ENU坐标系，基于地图的ND T输出的位置也是这个地图坐标系（ENU），如果你的场景非常大，比如一个横跨一个大城市，会有偏差（建议换成UTM系）。如果场景小，基本可以忽略。 你说的这个与ENU坐标系的偏移，坐标系偏差等，可以详细截图给我看看，你的描述需要更细节和清晰一点，我才能帮助解决问题。

luameows, 2023-03-17:
非常感谢您的答复。
以建图初始点为起点，先把rtk转到enu坐标系下，在建图时用rtk和lidar融合进行构建地图，随后我用同样的数据包和构建好的地图进行ndt匹配并记录位姿。将rtk->enu的轨迹和ndt->map的轨迹绘制出来发现，两者轨迹存在些偏差。如下图灰色是lidar轨迹，蓝色是rtk轨迹，大致重叠
[image]

还有个问题想请教下，在完成lidar+rtk融合建图后，是否可以认为第一帧关键帧的位姿就是地图坐标系和enu坐标系的偏差？我用lidar构建了一个未融合的地图，通过ndt->map绘制lidar轨迹后，乘以这个坐标系偏差后，轨迹大致能与enu对齐，但是精度上还是有些差异。如下图灰色是lidar轨迹，蓝色是rtk轨迹，红色是lidar->enu之后轨迹。
[image]
局部放大之后，这个偏差就比较大，想请教下这种一般是什么原因导致的呢？
[image]

JokerJohn, 2023-03-17:
第一个问题，RTK+lidar建图。
起点RTK质量如何，是否fix？ 起点的GNSS坐标严重影响WGS84->ENU的坐标转换精度，影响后续的RTK坐标数值以及最终优化。
你整段轨迹的RTK质量如何，是否存在明显的抖动情况或者局部比较差的情况？如果几乎全程都是RTK fix，那么你需要更相信RTK观测，重新调整RTK观测的加入频率以及协方差阈值，最终建图轨迹就整体偏向于更接近RTK轨迹，但比RTK轨迹更平滑（靠LIO）。另外你这个RTK轨迹是如何可视化的，利用repo中的simpleGNSS O dometry吗？如果是这样，这个simpleGNSS Odometry只考虑了相邻姿态的yaw角作为rotation，为平面运动设计的，比较粗略，只作参考。

JokerJohn, 2023-03-17:
第二个问题，第一帧关键帧的位姿就是地图坐标系和enu坐标系的偏差，这个是不准确的。
你当然可以把第一帧关键帧的位姿作为地图坐标系和enu坐标系的变换，但是不准确。因为
（1）你得到的第一针位姿不一定是准确的。
（2）RTK轨迹与LIO轨迹，两者其实是一个对齐的问题。轨迹对齐当然不只是依赖其中一个位姿。

luameows, 2023-03-18:
> 

（1）（2）rtk的质量应该还算ok，我们用华测的设备整个状态下rtk都是稳定解定位定向状态。 我试过直接把noise调小0.02(rtk精度2cm)，并提高rtk频率为50hz，发现融合后地图还有重影出现。感觉应该是rtk的质量不行？
（3）是的，参考repo的simplegnss，translation直接通过lla->enu获取，rotation则是根据rtk输出的车辆的yaw、pitch、yaw转换得到。我的场景跨度区域比较小，近平面的运动应该可以满足所需精度。
（4）还想请教下，如果我想在基于融合后的地图进行定位时，让rtk的姿态和ndt_mapping的姿态尽可能一致，还有什么需要注意的点么？

JokerJohn, 2023-03-18:
如果是这种情况，RTK质量较好，需要调高其权重。yaml文件中设置选取gnss点的距离，比如设置0.01m.。你这个提高RTK频率是什么意思，提高华测的RTK输出频率？？
mapOptimizationGps.cpp中 gps factor的noise model赋值时，GNSS协方差*10e-2等，可以调高GNSS权重，系统会更相信GNSS观测。另外我不知道你的重影区域区域是哪里，最好有示意图，你直接拿rtk位姿和雷达点云拼图也可能会有重影。

luameows, 2023-03-18:
> 如果是这种情况，RTK质量较好，需要调高其权重。yaml文件中设置选取gnss点的距离，比如设置0.01m.。你这个提高RTK频率是什么意思，提高华测的RTK输出频率？？ mapOptimizationGps.cpp中 gps factor的noise model赋值时，GNSS协方差*10e-2等，可以调高GNSS权重，系统会更相信GNSS观测。另外我不知道你的重影区域区域是哪里，最好有示意图，你直接拿rtk位姿和雷达点云拼图也可能会有重影。
> 
> 如果想在定位时让TRTK和ndt 姿态一致，直接引入GNSS约束进行定位的优化即可。

（1）是的，设置rtk输出频率为50hz，同时对gps的noise_mode赋值成0.02，调整gnss点选取距离0.01之后，建立的地图相比于设置(rtk的noise为0.2）会存在更大的重影。这个应该就是rtk的信号不好导致的吧？
[image]

（3）您的意思是，localization时把rtk作为观测去优化ndt的结果得到最终的位姿么？按照我理解，如果rtk信号足够多、质量都比较稳定，融合得到的map是enu坐标系下。此时基于地图的ndt得到的位姿应该和rtk得到的位姿即是一致的吧，无需再考虑lidar和rtk设备的外参。

JokerJohn, 2023-03-18:
这个问题已经回答过。
你用loam scan2map+rtk建图，本来就存在误差。

你用ndt算法在一个有误差的地图上定位，ndt算法正常情况下精度就比loam这种要低，这也存在误差。

这几个原因堆积起来，最终定位轨迹和rtk轨迹有差距，不是很正常吗？

closed 2023-03-18 by JokerJohn

luameows, 2023-03-18:
大概明白您说的了，我再消化下，非常感谢您的耐心解答。