JokerJohn/LIO_SAM_6AXIS #49: Does not work on Jetson

curiousityy, 2023-07-06:
Hi @JokerJohn,

Thank you for the great work. I am currently working with liosam 6 axis for my jetson platform (Jetson xavier NX). I Got the setup working on my PC (Ubuntu 20.04). When I port the same onto jetson (as a docker image). my map starts to drift and I continuously get a large velocity warning. Could you please let me know what could be wrong? I know it is to do with the params.yaml file. I tried reducing the voxel grid size as well, but it doesn't fix the issue.

Best Regards,
Sai

JokerJohn, 2023-07-06:
Based on your description, I suspect that the issue lies with the LiDAR-IMU extrinsic parameters. You may need to recalibrate the extrinsic parameters (mainly the rotation issue). The Docker version in the repository may not be entirely consistent with the newer versions, so we recommend directly compiling from the source code.

> Hi @JokerJohn,
>
> Thank you for the great work. I am currently working with liosam 6 axis for my jetson platform (Jetson xavier NX). I Got the setup working on my PC (Ubuntu 20.04). When I port the [...]

Based on your description, I suspect that the issue lies with the LiDAR-IMU extrinsic parameters. You may need to recalibrate the extrinsic parameters (mainly the rotation issue). The Docker version in the repository may not be entirely consistent with the newer versions, so we recommend directly compiling from the source code.

curiousityy, 2023-07-07:
Thank you for the update. But, if the issue was with the extrinsics, it shouldn't have worked in my PC as well right? It just doesn't work on the ARM64 platform. For the docker image, I created my own. I tried it directly on the jetson without the docker as well, but the issue still persists.

JokerJohn, 2023-07-07:
> Thank you for the update. But, if the issue was with the extrinsics, it shouldn't have worked in my PC as well right? It just doesn't work on the ARM64 platform. For the docker image, I created my own. I tried it directly on the jetson without the docker as well, but the issue still persists.

@curiousityy yes, you are right. I did not not tested on  jetson before, but if you can give some detailed error log. maybe I can help to solve this problem.

curiousityy, 2023-07-10:
@JokerJohn, please find the logs and the map drift (in the image).

Logs:

[ INFO] [1688718933.641949904]: ----> Feature Extraction Started.
[ INFO] [1688718933.699395248]: ----> Image Projection Started.
SAVE DIR:/media/anavs/SSD_2TB1/Home_Dec_7_jetson/lio-sam-6axis/catkin_ws/src/LIO_SAM_6AXIS/LIO-SAM-6AXIS/data/
[ INFO] [1688718934.174073936]: ----> IMU Preintegration Started.
[ WARN] [1688718960.243215888]: Large velocity, reset IMU-preintegration!
resetting graph
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ WARN] [1688718961.950581360]: Large velocity, reset IMU-preintegration!
resetting graph
[ WARN] [1688718965.039355248]: Large velocity, reset IMU-preintegration!
resetting graph
[ WARN] [1688718966.391134992]: Large velocity, reset IMU-preintegration!
resetting graph
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow

[image]

curiousityy, 2023-07-11:
Hi @JokerJohn, the bag files I have were recorded on the jetson. I have the lidar and the imu connected to the jetson and these recordings are tested on the PC. The same bag file when tested on the jetson itself don't work.