JokerJohn/LIO_SAM_6AXIS #52: Dataset availability through dropbox/google drive

juliangaal, 2023-07-27:
I am trying to access the hkust...full dataset (used for main .gif file?) to test some changes to the gps tools. I was wondering if the link to google drive was intentional, when the label is "dropbox"? I sent an access request through google drive a couple weeks ago

JokerJohn, 2023-07-27:
> I am trying to access the hkust...full dataset (used for main .gif file?) to test some changes to the gps tools. I was wondering if the link to google drive was intentional, when the label is "dropbox"? I sent an access request through google drive a couple weeks ago

There's a typo; I just forgot to edit it. However, Google Drive doesn't require permissions; you can download directly.
[image]

juliangaal, 2023-07-27:
That's strange. Using this link

[image]

I arrive straight at

[image]

Could you check the settings?

JokerJohn, 2023-07-28:
> That's strange. Using this link
> 
> [image]
> 
> I arrive straight at
> 
> [image]
> 
> Could you check the settings?

here is a one drive link.  [Download](https://hkustconnect-my.sharepoint.com/:u:/g/personal/xhubd_connect_ust_hk/Eea0g76yXehDt4asNoL6l14B6lEIX4eVX6aSG-CsnVUCjg?e=N2cSZx)

juliangaal, 2023-07-31:
worked like a charm, thank you!

closed 2023-07-31 by juliangaal