JokerJohn/LIO_SAM_6AXIS #53: simpleGpsOdom  关于 error origin ?

sywinyin, 2023-08-01:
感谢作者。
[image]
simpleGpsOdom.cpp第66行，或运算的三个条件都是一样的，  另外，为何这样子就是Error Origin ?

JokerJohn, 2023-08-01:
> 感谢作者。 [image] simpleGpsOdom.cpp第66行，或运算的三个条件都是一样的，有没有问题？ 另外，为何这样子判断就是Error Origin ?
如果起点ENU坐标太大，说明起点和坐标原点距离太远了（10km），球面当作平面处理误差极大了。建议重新设定ENU原点。

sywinyin, 2023-08-01:
> > 感谢作者。 [image] simpleGpsOdom.cpp第66行，或运算的三个条件都是一样的，有没有问题？ 另外，为何这样子判断就是Error Origin ?
> 
> 如果起点ENU坐标太大，说明起点和坐标原点距离太远了（10km），球面当作平面处理的话，z的误差就会很大。建议重新设定ENU原点。

感谢您及时的回复。
那么为什么，要使用三个一样的条件呢？
    if (abs(enu.x()) > 10000 || abs(enu.x()) > 10000 || abs(enu.x()) > 10000)

JokerJohn, 2023-08-01:
> > > 感谢作者。 [image] simpleGpsOdom.cpp第66行，或运算的三个条件都是一样的，有没有问题？ 另外，为何这样子判断就是Error Origin ?
> > 
> > 
> > 如果起点ENU坐标太大，说明起点和坐标原点距离太远了（10km），球面当作平面处理的话，z的误差就会很大。建议重新设定ENU原点。
> 
> 感谢您及时的回复。 那么为什么，要使用三个一样的条件呢？ if (abs(enu.x()) > 10000 || abs(enu.x()) > 10000 || abs(enu.x()) > 10000)

随便设置的工程阈值，一般不会超过这个数值，不必纠结。

closed 2023-08-01 by JokerJohn

sywinyin, 2023-08-01:
> > > > 感谢作者。 [image] simpleGpsOdom.cpp第66行，或运算的三个条件都是一样的，有没有问题？ 另外，为何这样子判断就是Error Origin ?
> > > 
> > > 
> > > 如果起点ENU坐标太大，说明起点和坐标原点距离太远了（10km），球面当作平面处理的话，z的误差就会很大。建议重新设定ENU原点。
> > 
> > 
> > 感谢您及时的回复。 那么为什么，要使用三个一样的条件呢？ if (abs(enu.x()) > 10000 || abs(enu.x()) > 10000 || abs(enu.x()) > 10000)
> 
> 随便设置的工程阈值，一般不会超过这个数值，不必纠结。

随便设置的阈值，我理解的。
我的意思，为什么不是  if (abs(enu.x()) > 10000 || abs(enu**.y()**) > 10000 || abs(enu.**z()**) > 10000)

JokerJohn, 2023-08-01:
> 随便设置的阈值，我理解的。 我的意思，为什么不是 if (abs(enu.x()) > 10000 || abs(**enu.y()**) > 10000 || abs(enu.**z()**) > 10000)

明白你的意思了，这个是笔误了，下波更新再改吧。

sywinyin, 2023-08-01:
> 明白你的意思了，这个是笔误了，下波更新再改吧。

OK ，感谢您。 另外，想问一下的是， 原点是在 *.launch里 这里设置的吗？
<node pkg="rostopic" type="rostopic" name="fake_gps_fix"
          args="pub gps/fix sensor_msgs/NavSatFix '{ header: auto, latitude: 22.3387279108228, longitude: 114.263622199692, altitude:  87.7310562}'"
          output="screen"/>

JokerJohn, 2023-08-01:
> OK ，感谢您。 另外，想问一下的是， 原点是在 *.launch里 这里设置的吗？ [image]

对，launch中修改原点是为了rviz-satellite插件可视化时候的正确显示。

sywinyin, 2023-08-01:
> 对，launch中修改原点是为了rviz-satellite插件可视化时候的正确显示。

我运行的campus_large_dataset.bag， if (abs(enu.x()) > 10000 || abs(enu.x()) > 10000 || abs(enu.x()) > 10000)    这个条件被激活了， 我应该怎么做呢， 是要在launch里修改原点吗？

JokerJohn, 2023-08-01:
> 我运行的campus_large_dataset.bag， if (abs(enu.x()) > 10000 || abs(enu.x()) > 10000 || abs(enu.x()) > 10000) 这个条件被激活了， 我应该怎么做呢， 是要在launch里修改原点吗？

launch里修改原点不会影响算法。两个办法。
1. 如果你确保转换后的ENU坐标无误，可以注释掉这几行代码。
2. bag包播放的起点选择GNSS状态比较好的位置，比如RTK fix区域。

sywinyin, 2023-08-01:
> launch里修改原点不会影响算法。两个办法。
>
> 1. 如果你确保转换后的ENU坐标无误，可以注释掉这几行代码。
> 2. bag包播放的起点选择GNSS状态比较好的位置，比如RTK fix区域。

第1点试过了，地图乱飞。 第2点还没试。
谢谢您。