JokerJohn/LIO_SAM_6AXIS #56: can not tun with exp11_lower_gallery.bag

lucianzhong, 2023-10-18:
roslaunch lio_sam_6axis hilti_pandar.launch

rosbag play exp11_lower_gallery.bag

JokerJohn, 2023-10-18:
Have you tried other data sequences in the HLITI dataset, and encountered the same issue? If so, it's highly likely that there's a problem with the LIDAR IMU extrinsic parameters setting. Please ensure that after the extrinsic transformation, the coordinate axes of the IMU align with those of the LIDAR. You may refer to these issues for further guidance. [issue20](https://github.com/TixiaoShan/LIO-SAM/issues/20)

lucianzhong, 2023-10-18:
I have not tried other HLITI dataset. Probably the extrinsic is wrong. Which HLITI dataset have you successfully tried ? 09 is ok or not?

lucianzhong, 2023-10-18:
By the way, can lio_sam give accurate height and pitch angle ?   I have tried an up_slope from underground park lot,then down slope into this park lot, the height and accuracy are not accurate until close loop detected.

JokerJohn, 2023-10-18:
In fact, about a year ago, I conducted tests on many sequences within the HILTI2022 dataset and confirmed that there was no issue with the extrinsic parameters setting. Recently, I haven't tried it and am not sure whether the preset extrinsics in the code I provided remain consistent with the original. It's worth noting that the HILTI dataset utilizes a PandarXT32 for LiDAR, which has a different coordinate system definition compared to Velodyne/Ouster, etc. If I remember correctly, it's front x, left y, which is 90 degrees different from the traditional coordinate system. You might want to verify that, and you can refer to this issue for further details [ISSUE4](https://github.com/Hilti-Research/hilti-slam-challenge-2022/issues/4) Additionally, I recommend that you use the relatively simpler data sequences exp01, exp02, exp03, exp04, and exp05.
[image]

lucianzhong, 2023-10-18:
Thanks a lot. 

Please have a look

> By the way, can lio_sam give accurate height and pitch angle ? I have tried an up_slope from underground park lot,then down slope into this park lot, the height and pitch angle are not accurate until close loop detected.

JokerJohn, 2023-10-18:
> By the way, can lio_sam give accurate height and pitch angle ? I have tried an up_slope from underground park lot,then down slope into this park lot, the height and pitch angle are not accurate until close loop detected.

I believe there might be some difficulty. LIO-SAM-6AXIS often performs well on ground robots since its initialization only requires a 6-axis IMU. In situations like going up and down stairs, there might be a lack of features which could lead to failure or a significant decrease in accuracy. I suggest you try using FAST-LIO2 which is based on the direct method; the filter framework might be more robust for such scenarios. Coupled with a reasonable loop closure (similar to LIO-SAM), and by filtering out the loops at the staircases through height filtering, this issue should be able to be resolved.

lucianzhong, 2023-10-18:
Got you. Thanks for the replay.

closed 2023-10-19 by lucianzhong