JokerJohn/LIO_SAM_6AXIS #57: gnns 初始化结束了，但是无法运行 terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'

zhao-zhibo, 2023-11-03:
我在适配我自己的数据集时，出现以下错误，这个该怎么进行处理呢？提示gtsam::IndeterminantLinearSystemException优化出错。
[image]
When I was adapting my own data set, the following error occurred. How should I deal with this? Prompt gtsam::IndeterminantLinearSystemException optimization error.

JokerJohn, 2023-11-03:
The cov of your gnss measurements can not be 0, since you need to set it as the noise model for your GPS factor.

zhao-zhibo, 2023-11-03:
> The cov of your gnss measurements can not be 0, since you need to set it as the noise model for your GPS factor.

因为我没有这个cov的参数，那是不是说我需要手动给我的gps的cov赋值呢？或者说我将gyaml文件中的psCovThreshold中的参数设置为负数，但是现在的结果两条轨迹是对不齐的，如下图所示。
Because I don’t have the parameters for this cov, does that mean I need to manually assign a value to the cov of my GPS? Or I set the parameter in psCovThreshold in the gyaml file to a negative number, but now the two trajectories are misaligned, as shown in the figure below.
[image]

JokerJohn, 2023-11-03:
轨迹没对齐的原因是GNSS factor没生效。gpsCovThreshold阈值的含义是筛选斜方差小于阈值的GNSS观测来构建GPS factor，你设置为负数，数值上没意义，更重要的是不会有Gps factor加入进来，相当于纯LIO了。没有cov这个结果的话，可以设置把noise_x, noise_y, noise_z为常数，比如0.01。但这意味着你认为每个GNSS观测都非常可信。
[image]

zhao-zhibo, 2023-11-04:
> 轨迹没对齐的原因是GNSS factor没生效。gpsCovThreshold阈值的含义是筛选斜方差小于阈值的GNSS观测来构建GPS factor，你设置为负数，数值上没意义，更重要的是不会有Gps factor加入进来，相当于纯LIO了。没有cov这个结果的话，可以设置把noise_x, noise_y, noise_z为常数，比如0.01。但这意味着你认为每个GNSS观测都非常可信。 [image]

非常感谢，已经按照你说的，将gnss的三个方向的噪声设置为0.01，现在轨迹对齐了。后面再仔细阅读一下您的代码部分，仔细学习一下。

closed 2023-11-04 by JokerJohn

zhao-zhibo, 2023-11-06:
> 轨迹没对齐的原因是GNSS factor没生效。gpsCovThreshold阈值的含义是筛选斜方差小于阈值的GNSS观测来构建GPS factor，你设置为负数，数值上没意义，更重要的是不会有Gps factor加入进来，相当于纯LIO了。没有cov这个结果的话，可以设置把noise_x, noise_y, noise_z为常数，比如0.01。但这意味着你认为每个GNSS观测都非常可信。 [image]

您好，不知道是否方便，加一下微信，然后更方便的进行交流呢？我的微信是beyond6557

JokerJohn, 2023-11-07:
建议issue下讨论，这样有同样问题的朋友也能看到，避免重复回答问题。

zhao-zhibo, 2023-11-07:
> 建议issue下讨论，这样有同样问题的朋友也能看到，避免重复回答问题。

好滴，谢谢回复，您的回复速度非常高，十分感谢