JokerJohn/LIO_SAM_6AXIS #67: 我用单雷达和双雷达拼接的建图效果不同，单雷达拐弯的时候位置不飘，但是建的图墙体是错的

zhao-zhibo, 2023-12-15:
非常感谢您的工作和之前对我的回答。我用的是雷神的32线设备，数据场景是在港口。使用单雷达建图效果和轨迹如下图所示：
[image]
双雷达是两个雷达根据时间戳和外参，然后做的点云拼接，拼接之后再跑代码结果如下图：
[image]
很显然，单雷达在拐弯的位置建的图直接飘了，但是位置是给修正过来了，这个是什么问题呢？

JokerJohn, 2023-12-15:
@zhao-zhibo 是否用了GpS？单雷达飞了，双雷达OK，很明显feature问题，打印下feature点数量看看，每一帧平面点多少，边缘点多少。

zhao-zhibo, 2023-12-15:
@JokerJohn 都用了gps，而且认为gps的质量非常高，噪声设置为0.01了，那我打印一下feature的数量。请问为什么关注gps呢？

JokerJohn, 2023-12-15:
@zhao-zhibo 很显然GPS只能约束位置，但存在一些数据不稳的情况，比如抖动，你都设置的0.01，说明你认为每个GPS观测都很准，这是不符合实际情况的。PGO会根据权重来优化，当优化结果与真实pose有些差距的时候，也可能导致飞

zhao-zhibo, 2023-12-15:
@JokerJohn PGO指的是什么意思呢？那我目前轨迹的情况是因为GPS给拉回来了对吧，也就是在拐弯的位置单雷达的效果是有问题的，然后gps直接拉回来，然后导致了轨迹是还行的，只是LIO建的图很差，我可以这样理解吗？