JokerJohn/LIO_SAM_6AXIS #68: 运行catkin build 报错！求助

JiangWeiHn, 2023-12-23:
我运行catkin build时报错如下：
---------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/noetic
Workspace:                   /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS
---------------------------------------------------------------------------------------
Build Space:        [exists] /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/build
Devel Space:        [exists] /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/devel
Install Space:      [unused] /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/install
Log Space:          [exists] /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/logs
Source Space:       [exists] /home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
---------------------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------------------
[build] No packages were found in the source space '/home/slamer/workspace/LIO_SAM_6AXIS/LIO-SAM-6AXIS/src'
[build] No packages to be built.
[build] Package table is up to date.                                                                                                                                                                                                                                                                                          
[build] Summary: All 0 packages succeeded!                                                                                                                                                                                                                                                                                    
[build]   Ignored:   None.                                                                                                                                                                                                                                                                                                    
[build]   Warnings:  None.                                                                                                                                                                                                                                                                                                    
[build]   Abandoned: None.                                                                                                                                                                                                                                                                                                    
[build]   Failed:    None.                                                                                                                                                                                                                                                                                                    
[build] Runtime: 0.0 seconds total.
这是怎么回事？

JiangWeiHn, 2023-12-25:
试了2个电脑，按照您的说明，安装，都是这个问题。

JokerJohn, 2023-12-26:
看你的路径，/home/slamer/workspace/LIO_SAM_6AXIS，你的包都没放在工作空间src目录下，编译当然显示空包啊

closed 2023-12-26 by JokerJohn