JokerJohn/LIO_SAM_6AXIS #72: gnss的杆臂

JiangWeiHn, 2024-03-05:
我的小车激光雷达和rtk的前后距离近1米，和imu的前后距离约半米，imu在中间，请问在加入gps信号的时候是否要考虑修正gps的杆臂效应？程序的中心是转换到激光雷达的坐标下还是imu的坐标下，这点不太清楚，请指教，谢谢！！

JokerJohn, 2024-03-07:
@JiangWeiHn 整个PGO优化出来的位姿都是在LIDAR坐标系。加入GPS观测时，你可以将GPS ENU坐标转换到IMU坐标系，再通过LIDAR- IMU外参转换到LIDAR系上。

JiangWeiHn, 2024-03-07:
收到，谢谢

hongSS0919, 2024-03-26:
> @JiangWeiHn 整个PGO优化出来的位姿都是在LIDAR坐标系。加入GPS观测时，你可以将GPS ENU坐标转换到IMU坐标系，再通过LIDAR- IMU外参转换到LIDAR系上。

这个要如何将GPS ENU坐标转换到IMU坐标系呢？

JiangWeiHn, 2024-04-02:
初始化的updateinitialGuess函数中也要加入转换程序，我忘记加了，所以跑飞了。改了就可以了

closed 2024-04-02 by JiangWeiHn

Jcs0213, 2025-01-02:
> 初始化的updateinitialGuess函数中也要加入转换程序，我忘记加了，所以跑飞了。改了就可以了

你好，想问一下您加入杆臂补偿后，融合RTK/GNSS的系统精度有没有显著提升呢

JiangWeiHn, 2025-01-02:
> > 初始化的updateinitialGuess函数中也要加入转换程序，我忘记加了，所以跑飞了。改了就可以了
> 
> 你好，想问一下您加入杆臂补偿后，融合RTK/GNSS的系统精度有没有显著提升呢

有显著提升，对于我来说

G4419, 2025-01-13:
> hisGPS.pose.pose.

没太看懂什么意思？thisGPS.pose.pose.orientation不是从simpleGpsOdom获得的吗？为什么说是从IMU预积分的地方获得的

G4419, 2025-01-13:
> 先建立gnss到lidar的pose: gtsam::Pose3 gnss2Lidar =gtsam::Pose3(gtsam::Rot3(1, 0, 0, 0),gtsam::Point3(-0.45328,-0.3,0.21647)) 然后根据当前的gnss坐标，转换到lidar的坐标，这个是在addGPSFactor（）函数中： double r_w,r_x,r_y,r_z; 姿态是我从IMU预积分的地方获得的。 r_w = thisGPS.pose.pose.orientation.w; r_x = thisGPS.pose.pose.orientation.x; r_y = thisGPS.pose.pose.orientation.y; r_z = thisGPS.pose.pose.orientation.z; gtsam::Pose3 gpsPose = gtsam::Pose3(gtsam::Rot3::Quaternion(r_w, r_x, r_y, r_z), gtsam::Point3(gps_x, gps_y, gps_z)); gtsam::Pose3 gpsInLidarPose = gpsPose.compose(gnss2Lidar); 我不知道这样写哪不对，运行数据集，小车直接飘飞。请作者帮忙看下，谢谢！！

没太看懂什么意思，thisGPS.pose.pose.orientation不是从simpleGpsOdom获得的吗？为什么说是从IMU预积分的地方获得的

G4419, 2025-01-13:
> @JiangWeiHn 整个PGO优化出来的位姿都是在LIDAR坐标系。加入GPS观测时，你可以将GPS ENU坐标转换到IMU坐标系，再通过LIDAR- IMU外参转换到LIDAR系上。
意思是P_ENU_gps->P_ENU_imu->P_ENU_lidar，这样的转换吗?