JokerJohn/LIO_SAM_6AXIS #74: [lio_sam_6axis_imageProjection-2] process has died

pedrotomas27, 2024-03-18:
Hello there, i am pretty new to ROS and right now i am trying to implement your method to different types of LiDARs, at the moment im trying to implement it with data from a velodyne m1600, that is a solid state, i've already changed the code in order to accept the entry velodyne_m1600 and his data structure, but when launched i get this message 
[lio_sam_6axis_imageProjection-2] process has died [pid 55405, exit code -11, cmd /home/pedro/catkin_ws/devel/lib/lio_sam_6axis/lio_sam_6axis_imageProjection __name:=lio_sam_6axis_imageProjection __log:=/home/pedro/.ros/log/9f001806-e544-11ee-903c-a33d96ef37be/lio_sam_6axis_imageProjection-2.log].
log file: /home/pedro/.ros/log/9f001806-e544-11ee-903c-a33d96ef37be/lio_sam_6axis_imageProjection-2*.log

i dont think this is made by my changes tho...

Heres my topics
[image]

And here is my config file
[image]
[image]

Does anyone have any idea whats the problem?

JokerJohn, 2024-03-19:
@pedrotomas27 I've noticed from your log that the lio_sam_6axis_imageProjection node is crashing, which is linked to the issues you're encountering upon adding Velodyne solid-state LiDAR data structures. This repository does not support solid-state LiDAR because the lio_sam_6axis_imageProjection node projects the point cloud into an image based on beams and resolution. With solid-state LiDAR, it's unclear how to manage the index of image points after projection. Addressing this might meet your needs; otherwise, I'd advise against further modifications to this repository.

Additionally, I observed that the IMU noise and bias settings in your configuration file are set very low. I would not recommend this approach, as it implies high trust in your IMU measurements. In a stable LIO system, the focus is generally on LiDAR data. I suggest starting with debug settings of 0.01/0.1 for the noise and bias.

pedrotomas27, 2024-03-22:
Hello again!
Thank you for your feedback @JokerJohn!
I found out that my error was caused by calling the same function twice in the imageProjection.cpp file...
I actually manage to make the method work with velodyne M1600, i will make a pull request adding those changes later!

closed 2024-03-22 by pedrotomas27