JokerJohn/LIO_SAM_6AXIS #78: exp01_construction_ground_level 数据集报 Large velocity, reset IMU-preintegration!

sywinyin, 2024-03-29:
您好。
我使用 hilti2022的exp01_construction_ground_level 数据集做测试。
按照您这篇提示https://github.com/JokerJohn/LIO_SAM_6AXIS/blob/main/doc/adaption.md，参数设置为：
[image]
结果报错
[ WARN] [1711695952.247137334]: Large velocity, reset IMU-preintegration!
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

[image]

请问什么原因呢？

JokerJohn, 2024-03-29:
@sywinyin 这个其实是预积分节点不稳定带来的问题。预积分节点接收上一时刻优化后的雷达位姿，通过预积分约束，然后用来修正imu bias，利用新优化的bias和新雷达时间戳之后的新imu数据去做预测，给下一针雷达匹配做初始位姿。同时也会用于投影节点的点云畸变去除。从这个流程看，liosam前后端耦合非常紧密。而其核心的预积分节点，直接使用gtsam imu factor，当你噪声模型设置不对，或者外步运动剧烈的时候，这个节点很容易因为病态优化而挂掉，导致整个系统崩溃。
看你的报错说明系统欠定，有些参数解不出来。我猜测是exp01应该是手持采集的，在起始阶段运动幅度过大导致的问题，你换一个运动比较平缓的数据，比如exp05，可能就好点。

sywinyin, 2024-03-29:
@JokerJohn   exp05在下载中。 请允许我在这里插入一个别的问题：
代码1174行和1178行为什么要注释呢，这样就把noise_x，noise_y，noise_z 强制重写了啊。
这是为什么呢？
[image]

sywinyin, 2024-04-01:
@JokerJohn  感谢及时的回复！
使用hilti2022的数据集 exp05_construction_upper_level_2.bag ， 参数同上exp01的设置。
[image]
该exp05数据是没有gps数据的。
**结果仍然报错：**
[ WARN] [1711934855.336546640]: Large velocity, reset IMU-preintegration!
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

您可以将您运行成功的exp01的参数设置发我一下吗，我的邮箱是 sywin_yin@163.com