JokerJohn/LIO_SAM_6AXIS #82: gps原点问题

wolf943134497, 2024-04-26:
Hi @JokerJohn  Thanks for your great works!
您好，关于gps原点问题有几个问题想请教一下。
1.  当前版本代码跑出来lio pose与gps GT算ape，发现误差主要在初始阶段，误差在1-6m量级，请问这是什么原因？
2.  我将simpleGpsOdom.cpp 和 mapOptizationGps.cpp中gps原点设置成一致，发现ape误差也只在初始阶段比较大
3. 我有几个bag，是在相邻街道采集的，为了使所有bag建出来的地图有统一的gps原点，在代码中我将gps原点写成一个固定值(为其中一个bag的第一帧gps)，但是这样跑发现lio会飞掉，想请教一下这可能是什么原因造成的？这几个bag的gps原点相近的。
4. 如果想实现多个bag统一gps原点建图，怎么修改代码比较好，请问有什么建议嘛？

谢谢

JokerJohn, 2024-04-27:
@wolf943134497 

在simpleGpsOdom.cpp里面，初始化的时候会把GNSS LLA原点发布出来。这个原点LLA就是GNSS的坐标原点。后续mapOptizationGps.cpp中用于融合的ENU观测点都是针对这个原点LLA得到的投影坐标，所以你mapOptizationGps中也不需要考虑LLA原点是多少。
[image]
如果你想把几个bag在同一个LLA原点上建图，你可以在simpleGpsOdom.cpp中，初始化时直接设置固定的LLA起点。比如我会写在构造函数里
```
GNSSOdom(ros::NodeHandle &_nh) {
        nh = _nh;
        gpsSub = nh.subscribe(gpsTopic, 1000, &GNSSOdom::GNSSCB, this,
                              ros::TransportHints().tcpNoDelay());
        left_odom_pub = nh.advertise<nav_msgs::Odometry>("/gps_odom", 100, false);
        init_origin_pub = nh.advertise<nav_msgs::Odometry>("/init_odom", 10000, false);
        left_path_pub = nh.advertise<nav_msgs::Path>("/gps_path", 100);
    }
geo_converter.Reset(lla[0], lla[1], lla[2]);  // 设置固定的LLA起点
initENU = true；
```

然后你注释掉GNSSCB中的这段代码即可。
```
if (!initENU) {
            ROS_INFO("Init Orgin GPS LLA  %f, %f, %f", msg->latitude, msg->longitude,
                     msg->altitude);
            geo_converter.Reset(lla[0], lla[1], lla[2]);
            initENU = true;

            /** publish initial pose from GNSS ENU Frame*/
            nav_msgs::Odometry init_msg;
            init_msg.header.stamp = msg->header.stamp;
            init_msg.header.frame_id = odometryFrame;
            init_msg.child_frame_id = "gps";
            init_msg.pose.pose.position.x = lla[0];
            init_msg.pose.pose.position.y = lla[1];
            init_msg.pose.pose.position.z = lla[2];
            init_msg.pose.covariance[0] = msg->position_covariance[0];
            init_msg.pose.covariance[7] = msg->position_covariance[4];
            init_msg.pose.covariance[14] = msg->position_covariance[8];
            init_msg.pose.pose.orientation = yaw_quat_left;
            init_origin_pub.publish(init_msg);
            return;
        }
```
这样就能保证你多个bag包在建图时使用的GNSS观测的ENU坐标在同一个起点上，经过PGO融合后，都会对齐到ENU系下。

updateInitialGuess函数这里固定住了起点(PGO起点)，强行与此时的GNSS观测固定到一起，这是为了适配HKUST- campus初始阶段GNSS很差做的处理，也防止优化出现ill-conditioned的情况。如果你初始GNSS质量还不错，可以去掉这个factor，因为这个可能是导致你的1.2点的核心原因。
[image]

wolf943134497, 2024-04-28:
@JokerJohn  Thanks for your rapid reply
非常感谢您的建议，问题解决。

closed 2024-04-28 by wolf943134497