JokerJohn/LIO_SAM_6AXIS #84: [lio_sam_6axis_mapOptmization-5] process has died, whenever I play the rosbag

shchoi00, 2024-05-22:
I appreciate for you job!

Using #70 I can build the package with noetic.

But, when I play the rosbag, it prints some error like following:

`[lio_sam_6axis_mapOptmization-5] process has died [pid 1468503, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/lio_sam_6axis/lio_sam_6axis_mapOptmization __name:=lio_sam_6axis_mapOptmization __log:=/home/ubuntu/.ros/log/a8830fe6-184d-11ef-b4f1-87907ca5256a/lio_sam_6axis_mapOptmization-5.log].`

I'm using ouster OS-1 64 and internal IMU(6 axis) as well.

Here is my yaml config. There are few changes of param.

io_sam_6axis:

```
  #  velodyne16 and stim300(6 axis)

  # Topics
  pointCloudTopic: "/ouster/points"               # Point cloud data
  imuTopic: "/ouster/imu"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "gps_odom"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useGPS: false
  gpsFrequence: 10                            # set proper gps frequence to align gps and lidar timestamp
  useImuHeadingInitialization: false         # if using GPS data, set to "false"
  useGpsElevation: true                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data

  # debug setting
  debugLidarTimestamp: false
  debugImu: false
  debugGps: false

  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # actually we do not use the floder to save maps

  # depends on your own lidar!!!!
  sensor: ouster                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox' or 'hesai'
  N_SCAN: 64                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 1024                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1
  lidarMinRange: 1.5                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0

  # IMU Settings
  imuFrequence: 100
  imuAccNoise: 0.0011501915187049582
  imuGyrNoise: 5.084312924828687e-05
  imuAccBiasN: 3.729854069064516e-05
  imuGyrBiasN: 6.509823412408377e-07
  imuGravity: 9.80511
  imuRPYWeight: 0.01
```

Additionally, When I play [FusionPortable](https://ram-lab.com/file/site/fusionportable/dataset/fusionportable/) DATASET which located in readme,

JokerJohn, 2024-05-22:
@shchoi00 show me your full yaml file, do not delete any parameters here

yunliuh, 2024-06-08:
I am encountering the same issue as you did. Have you managed to resolve it?

yunliuh, 2024-06-08:
I have found that the node crashes when attempting to add elements to surroundingKeyPosesDS (during the operation surroundingKeyPosesDS->push_back) or when destructing surroundingKeyPosesDS.

yunliuh, 2024-06-09:
I resolved the issue by following the approach provided in the link https://github.com/TixiaoShan/LIO-SAM/issues/191.

JokerJohn, 2024-11-25:
congrats！

closed 2024-11-25 by JokerJohn