JokerJohn/LIO_SAM_6AXIS #85: imuPreintegration崩溃

LiquoriceH, 2024-07-09:
[image]
请问这个是为什么呀，程序一开始就一直报

JokerJohn, 2024-07-09:
@LiquoriceH 我猜你需要检查下以下两个问题
1.Prior pose与imu的噪声设置，Prior pose噪声越小，代表你越信任LIDAR Pose.
2.IMU的外参方向有没有问题。
[image]

LiquoriceH, 2024-07-09:
> @LiquoriceH 我猜你需要检查下以下两个问题 1.Prior pose与imu的噪声设置，Prior pose噪声越小，代表你越信任LIDAR Pose. 2.IMU的外参方向有没有问题。 <img alt="image" width="985" src="https://private-user-images.githubusercontent.com/25702851/346856808-af08c77a-52d0-4790-8323-af9a1010da8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTE5NDQsIm5iZiI6MTcyMDUxMTY0NCwicGF0aCI6Ii8yNTcwMjg1MS8zNDY4NTY4MDgtYWYwOGM3N2EtNTJkMC00NzkwLTgzMjMtYWY5YTEwMTBkYThmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA3NTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxNDk4ZDliOWIwZjhhMDdiYzdiZGMwYmJmYTIwNmI5NjVhOWIxNjFhMjhhMDVlNGM3YWE3NjRmNzk2ZmQyNWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.t-HM2XHq4wKc_QwlskUa3QBEW8o9iD1PtvrY-GoKmpQ">

IMU外参矩阵我确认了没有问题，雷达我使用的MID360，IMU没有使用内置的，用的是一个外置的比较垃圾的IMU，您的意思是我应该增大IMU的Noise，更相信Lidar吗，这个一般Noise怎么调整呢，感觉一头雾水

JokerJohn, 2024-07-09:
@LiquoriceH 本算法并没有提供对mid360的直接支持，我不确定能否顺利跑通。至于，IMU的Noise这一部分，mid360 IMU噪声很大，我建议你不要动LiDAR pose噪声，调大IMU部分的噪声参数(noise，bias)，甚至可以设置为是1.0。

LiquoriceH, 2024-07-09:
> @LiquoriceH 本算法并没有提供对mid360的直接支持，我不确定能否顺利跑通。至于，IMU的Noise这一部分，mid360 IMU噪声很大，我建议你不要动LiDAR pose噪声，调大IMU部分的噪声参数(noise，bias)，甚至可以设置为是1.0。

感谢您的建议。MID360是我自己参考开源代码适配的，图中的那个问题是偶现的，并且出现的频率较高，没有出现的时候是可以建图的。
[image]
[image]
您说的“IMU部分的噪声参数(noise，bias)”是指上面这俩个参数吗。

LiquoriceH, 2024-07-09:
> @LiquoriceH 对，priorBiasNoise你可以调整到1e-2。此外，主要是imuAccbiasN, imuAccNoise,imuGyroNoise这几个可以调大，比如0.1/0.5/1.0，设置的时候注意noise一般比bias大一个数量级。 “图中的那个问题是偶现的”，这个根本原因还是GTSAM 预积分这部分代码实现得不好。 <img alt="image" width="940" src="https://private-user-images.githubusercontent.com/25702851/346875990-4a1bfa40-1f23-49bd-ab01-7708f737263d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MTM2NzcsIm5iZiI6MTcyMDUxMzM3NywicGF0aCI6Ii8yNTcwMjg1MS8zNDY4NzU5OTAtNGExYmZhNDAtMWYyMy00OWJkLWFiMDEtNzcwOGY3MzcyNjNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA5VDA4MjI1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3YjlmNjA2ZmZhOWIyYTRmYzkyZDIwODViNDllMDY4MThhZmRmYmJkYjM3MzA5OWZiYWMyYzJjM2IzYWJmYTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PMSqQSbrVzkxsu2EvCq7_CXF3jEAoL3QF4z-vLrPWA0">
> 
> 另外，如果你已经是配好了mid360 LIDAR部分的代码，可以提交合并到repo中。

好的，我去试一下调整这些参数。我整理好后会提交合并。感谢博士。