JokerJohn/LIO_SAM_6AXIS #89: 融合精度

LiquoriceH, 2024-09-13:
请问如果rtk和雷达之间存在较大的tf平移变换，会影响地图对齐的精度吗，即是否对齐后会存在固定偏差

LiquoriceH, 2024-10-17:
> @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。

博士，我想保证mapping的path和gps_odom尽可能地重合，所以我降低了rtk的方差，但是虽然轨迹重合了，但是建图会在某些情况下存在重影，我感觉可能是因为rtk的观测可能并不是特别的准。请问您有什么见解和建议吗

JokerJohn, 2024-10-30:
@LiquoriceH 你的看法没问题的，由于卫星分布的原因，GNSS观测的z轴误差基本都比XY大一个数量级。在位姿图融合的时候，其实是相当于用RTK的位置给待优化位姿施加位置先验(xyz)，旋转是没有约束的。结果就是优化后的轨迹会尽量往RTK轨迹上靠，有重影是正常的。我建议你严格限制加入加入的RTK观测，比如设置更小的cov阈值。此外，你可以把LIO理解成一个平滑器，它的主要问题是存在累积误差，你用比较准的RTK观测就能很好的消除，其实并不需要那么多的RTK观测，一小段轨迹上有少量就行。

G4419, 2025-01-13:
> @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。

请问下六轴IMU的朝向和ENU坐标系都不统一，这该怎么加呢？

LiquoriceH, 2025-01-13:
> > @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。
> 
> 直接加上两者之间的偏移，gtsam优化时会自动往GPS轨迹上拉，可以这么理解吗？

我是通过得到了RTK的ENU坐标后，通过RTK主线天和Lidar的TF关系，得到了Lidar的ENU坐标，然后反算LLA去加的观测

LiquoriceH, 2025-01-13:
> > @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。
> 
> 直接加上两者之间的偏移，gtsam优化时会自动往GPS轨迹上拉，可以这么理解吗？

我是通过计算RTK的ENU坐标后，然后通过RTK主天线和Lidar的TF关系，得到了Lidar的ENU坐标，最后反算LLA去加的观测

G4419, 2025-01-14:
> > > @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。
> > 
> > 
> > 直接加上两者之间的偏移，gtsam优化时会自动往GPS轨迹上拉，可以这么理解吗？
> 
> 我是通过计算RTK的ENU坐标后，然后通过RTK主天线和Lidar的TF关系，得到了Lidar的ENU坐标，最后反算LLA去加的观测

用的是九轴的IMU吗？不然RTK主天线和Lidar的TF关系是怎么得到的呢？

LiquoriceH, 2025-01-14:
> > > > @LiquoriceH 在GPS factor中将ENU下的GNSS观测，把这个偏移加上。
> > > 
> > > 
> > > 直接加上两者之间的偏移，gtsam优化时会自动往GPS轨迹上拉，可以这么理解吗？
> > 
> > 
> > 我是通过计算RTK的ENU坐标后，然后通过RTK主天线和Lidar的TF关系，得到了Lidar的ENU坐标，最后反算LLA去加的观测
> 
> 用的是九轴的IMU吗？不然RTK主天线和Lidar的TF关系是怎么得到的呢？

最简单最快的方法你直接去量一下，也可以用rtk和lidar这俩传感器进行联合标定，不需要IMU

JokerJohn, 2025-01-20:
@G4419 @LiquoriceH 

GTSAM新版本已经合并了GPSArm Factor，可以参考下[GPSFactor](https://github.com/borglab/gtsam/blob/717e3bea63a118fe17e3891b294191c0ae7d7002/gtsam/navigation/GPSFactor.cpp)

closed 2025-01-20 by JokerJohn

LiquoriceH, 2025-01-20:
> [@G4419](https://github.com/G4419) [@LiquoriceH](https://github.com/LiquoriceH)
> 
> GTSAM新版本已经合并了GPSArm Factor，可以参考下[GPSFactor](https://github.com/borglab/gtsam/blob/717e3bea63a118fe17e3891b294191c0ae7d7002/gtsam/navigation/GPSFactor.cpp)

确实方便了很多，之前都是手动转的

LiquoriceH, 2025-01-20:
还带了unknow arm版本的

JokerJohn, 2025-01-25:
> 还带了unknow arm版本的

@LiquoriceH 有杆臂数据的话可以提个PR，我合并一下。

LiquoriceH, 2025-01-26:
> > 还带了unknow arm版本的
> 
> [@LiquoriceH](https://github.com/LiquoriceH) 有杆臂数据的话可以提个PR，我合并一下。

博士是指带RTK数据的bag包嘛？

JokerJohn, 2025-01-27:
> > > 还带了unknow arm版本的
> > 
> > [@LiquoriceH](https://github.com/LiquoriceH) 有杆臂数据的话可以提个PR，我合并一下。
> 
> 博士是指带RTK数据的bag包嘛？

带杆臂数据的demo案例，以及整合的代码都可以