JokerJohn/LIO_SAM_6AXIS #90: GPS的位置是归算到LiDAR的位置上吗？有提供相应的接口吗？

zhao-zhibo, 2024-09-18:
LIO-SAM中extrinsicRot、extQRPY和extrinsicTrans的这几个变量是描述imu的加速度计和陀螺仪与LiDAR的旋转和位置关系，而且liosam中的定位过程都是以lidar为中心求出其相对于世界坐标系的位置和姿态的。那GPS的位置和LiDAR也有一个位置偏移，gps的位置是否应该也要归算到LiDAR位置上呢，这个位置偏移是不是没有提供配置的接口呢？

JokerJohn, 2024-10-17:
@zhao-zhibo “GPS的位置和LiDAR也有一个位置偏移”，这个确实存在，只有平移，需要手动测量下。即使是RTK数据，z轴误差一般也高于x/y一个数量级(10cm)，手动测量多两厘米少两厘米的误差也挺常见，这意味着测量误差是厘米，手动标定误差也是厘米级，LIO精度也是厘米姐，多点少点我觉得差别不大。当然最好自行修正下，我这个代码里没有提供接口处理。

zhao-zhibo, 2024-10-17:
@JokerJohn 那这么说的话，gps的位置是在n系下，lidar坐标系是在其自身坐标系下。那需要自行修正的话，也就是说将gps的位置全部自行归算到lidar的位置上是吧？

JokerJohn, 2024-11-25:
@zhao-zhibo 可以这么理解。简单说，这里的SLAM和GPS融合，相当于有一些稀疏的GPS 位置点(只有3维)，然后有稳定的SLAM轨迹(差不多10Hz 6维)，基于GPS的优化就是把SLAM轨迹往GPS位置轨迹上对齐，使其残差最小化。这就意味着SLAM轨迹全部被拉到GNSS的ENU轨迹上，坐标系为绝对的ENU系，如果LiDAR和GNSS之间是一个稳定的刚体变换且你的点云地图后续用来定位用的话，我认为这个坐标系中❤️在雷达上还是GNSS上对地图没啥影响。

zhao-zhibo, 2024-11-26:
@JokerJohn 感谢你的积极解答，最后一句中"这个坐标系中在雷达上还是GNSS上对地图没啥影响是啥意思？"，是指最终的坐标系是在初始位置的雷达坐标系还是ENU坐标系对最终地图没影响吗？ 
另外一个问题：我看了另外一个issue https://github.com/JokerJohn/LIO_SAM_6AXIS/issues/50 ，是怎么保证LiDAR坐标系和ENU坐标系两个融合的时候让LI的轨迹向ENU坐标系上拉，而不是相信LiDAR坐标系呢？

JokerJohn, 2024-11-26:
这个很简单，你推一下GPS factor的雅可比矩阵就知道为什么了。我在[知乎解答过](有哪些GPS和Lidar SLAM融合的建图算法？ - Catalina的回答 - 知乎
https://www.zhihu.com/question/507251261/answer/2291977850)，不过有点乱，你只看雅可比矩阵相关的就行。
[image]

zhao-zhibo, 2024-11-26:
@JokerJohn  GPS factor的雅可比矩阵我推导过，[二维空间和三维空间刚体变换中雅克比矩阵的推导](https://blog.csdn.net/zzb714121/article/details/135006190?spm=1001.2014.3001.5501)。我又看了你的知乎链接，让我看gps因子的雅可比矩阵是想说明哪个问题呢？
另外，"GPS是先验，只约束一个node的位姿。LIO里的都是相对位姿约束，作用在两个node上"，先验就相当于直接告诉因子图对应的一个节点，它的位置是在这个坐标系了，这么说先验的作用还是蛮大的，这么理解应该没问题吧？

JokerJohn, 2024-11-26:
@zhao-zhibo 你都推过了，那肯定清楚 Gtsam中GPS factor的误差定义是(估计-观测)还是(观测-估计)了，优化往误差最小的方向走，观测的是ENU系的位置，估计的是LIO的位姿的平移部分。这就意味着优化的时候GNSS这些ENU点根据权重大小相当于固定住了，LIO的轨迹往ENU点上靠，从而实现平移误差最小。这不就是说把LIO轨迹往ENU点上拉吗？
SLAM是不可观的，除了第一个起点外，这后面的都是相对位姿，所以你可以任意选择起点，其相对位姿关系不会变化。一般实际中会先固定第一个点给一个先验，从而固定SLAM的坐标系，使得问题可以求解。这第一个先验位姿和GNSS的作用一样。

zhao-zhibo, 2024-11-26:
@JokerJohn 感谢你的详细回复，Gtsam中的gps factor的误差定位是估计-观测，我理解了你的意思了。

closed 2025-01-20 by JokerJohn