JokerJohn/LIO_SAM_6AXIS #92: 9改6问题

Iiguoren, 2024-10-17:
作者大大我看代码发现主要变动在imuconvert中对RPY改为单位矩阵，而在后面transformupdate中与磁力计数据按照权重融合时候并没有关掉这个融合，这个RPY的单位阵不会有影响吗
``
if (cloudInfo.imu_available == true)
        {
            if (std::abs(cloudInfo.imu_pitch_init) < 1.4)
            {
                double imuWeight = imuRPYWeight;
                tf2::Quaternion imuQuaternion;
                tf2::Quaternion transformQuaternion;
                double rollMid, pitchMid, yawMid;

                // slerp roll
                transformQuaternion.setRPY(transformTobeMapped[0], 0, 0);
                imuQuaternion.setRPY(cloudInfo.imu_roll_init, 0, 0);
                tf2::Matrix3x3(transformQuaternion.slerp(imuQuaternion, imuWeight)).getRPY(rollMid, pitchMid, yawMid);
                transformTobeMapped[0] = rollMid;

                // slerp pitch
                transformQuaternion.setRPY(0, transformTobeMapped[1], 0);
                imuQuaternion.setRPY(0, cloudInfo.imu_pitch_init, 0);
                tf2::Matrix3x3(transformQuaternion.slerp(imuQuaternion, imuWeight)).getRPY(rollMid, pitchMid, yawMid);
                transformTobeMapped[1] = pitchMid;
            }
        }
``

JokerJohn, 2024-10-17:
@Iiguoren 你说到点子上了，这部分确实会有影响，需要关闭，但除了初始化，其他时候差别不大，所以没太在意。因为松耦合的原因，IMU这里的用途只是做scan-to-map匹配的初值。欢迎测试后提交下代码，我直接合并进去。

cuilongfei1231, 2024-11-28:
您好，在哪里关掉您说的这个融合？？请问是那个cpp文件，我没有找到在那个位置修改，“ transformupdate中与磁力计数据按照权重融合时候并没有关掉这个融合，这个RPY的单位阵不会有影响吗”
``
if (cloudInfo.imu_available == true)
{
if (std::abs(cloudInfo.imu_pitch_init) < 1.4)
{
double imuWeight = imuRPYWeight;
tf2::Quaternion imuQuaternion;
tf2::Quaternion transformQuaternion;
double rollMid, pitchMid, yawMid;

            // slerp roll
            transformQuaternion.setRPY(transformTobeMapped[0], 0, 0);
            imuQuaternion.setRPY(cloudInfo.imu_roll_init, 0, 0);
            tf2::Matrix3x3(transformQuaternion.slerp(imuQuaternion, imuWeight)).getRPY(rollMid, pitchMid, yawMid);
            transformTobeMapped[0] = rollMid;

            // slerp pitch
            transformQuaternion.setRPY(0, transformTobeMapped[1], 0);
            imuQuaternion.setRPY(0, cloudInfo.imu_pitch_init, 0);
            tf2::Matrix3x3(transformQuaternion.slerp(imuQuaternion, imuWeight)).getRPY(rollMid, pitchMid, yawMid);
            transformTobeMapped[1] = pitchMid;
        }
    }