JokerJohn/LIO_SAM_6AXIS #103: Calibration

pedestrain123, 2025-03-22:
你好请问如何精确进行激光雷达与组合惯导（gnss/ins）之间的外参标定？

JokerJohn, 2025-03-22:
简单方法，保存一个LIO轨迹和INS轨迹，INS轨迹只保留很精确的点(协方差很小)。直接EVO对齐就能得到transform了，内部用的是Umeyama算法。此外，如果你是只是需要得到INS坐标系下的点云地图，可以不必标定，直接将GPS Factor换成PriorFactor<Pose3>即可，因为图优化的时候会自动往prior位置上靠以实现误差最小化，最后图优化的结果也就自然在INS坐标系下了。

pedestrain123, 2025-03-22:
这是一种基于手眼的标定方法，目前我也正在使用，因为组合惯导我们用的是华测630，输出的位姿精度相对较高。激光里程计用的是ndt omp，目前从标定精度上来看不是很稳定，数据采集是通过绕八字平面运动。激光雷达用的速腾m1固态雷达，有什么好的方法可以提高纯激光里程计精度吗？针对这种运动方式

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年03月22日 21:17 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [JokerJohn/LIO_SAM_6AXIS] Calibration (Issue #103) |

简单方法，保存一个LIO轨迹和INS轨迹，INS轨迹只保留很精确的点(协方差很小)。直接EVO对齐就能得到transform了，内部用的是Umeyama算法。此外，如果你是只是需要得到INS坐标系下的点云地图，可以不必标定，直接将GPS Factor换成PriorFactor即可，因为图优化的时候会自动往prior位置上靠以实现误差最小化，最后图优化的结果也就自然在INS坐标系下了。

—

简单方法，保存一个LIO轨迹和INS轨迹，INS轨迹只保留很精确的点(协方差很小)。直接EVO对齐就能得到transform了，内部用的是Umeyama算法。此外，如果你是只是需要得到INS坐标系下的点云地图，可以不必标定，直接将GPS Factor换成PriorFactor即可，因为图优化的时候会自动往prior位置上靠以实现误差最小化，最后图优化的结果也就自然在INS坐标系下了。

JokerJohn, 2025-03-22:
里程计我觉得用fastlio2就够了，ndt精度不太行的。标定方式可以参考下一些SLAM数据集中的做法。

pedestrain123, 2025-03-22:
但是我目前不知道lidar和imu之间的外参啊？是不是无法使用fastlio2这个方法？

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年03月22日 21:36 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [JokerJohn/LIO_SAM_6AXIS] Calibration (Issue #103) |

里程计我觉得用fastlio2就够了，ndt精度不太行的。标定方式可以参考下一些SLAM数据集中的做法。

—

里程计我觉得用fastlio2就够了，ndt精度不太行的。标定方式可以参考下一些SLAM数据集中的做法。

JokerJohn, 2025-04-27:
@pedestrain123 LiDAR IMU外参必须要。我建议通过相机中转，先标相机IMU，再标相机雷达，最后得到LIDAR IMU外参。