JokerJohn/LIO_SAM_6AXIS #105: 添加重定位功能

kkkkkk812, 2025-04-27:
你好，现在我想在您的代码基础上，添加icp的点云重定位的功能，请问您有什么思路吗？我思考的是我在mapOptmization.cpp文件中添加pcd点云地图的加载和icp的匹配功能就可以实现这个功能了，期待您的指导

JokerJohn, 2025-04-27:
@kkkkkk812 其实我不建议你这么做，LIO-SAM因为代码框架问题，目前只能做一些补丁。其鲁棒性和精度是很难喝当前一些SOTA的LIO方法相比的，与其改这个，你不如直接修改其他的算法，比如FAST-LIO2.

如果坚持要改LIO-SAM的话，其实github上已经有不少代码可以参考了，你看看这个 https://github.com/Gaochao-hit/LIO-SAM_based_relocalization

kkkkkk812, 2025-04-28:
> [@kkkkkk812](https://github.com/kkkkkk812) 其实我不建议你这么做，LIO-SAM因为代码框架问题，目前只能做一些补丁。其鲁棒性和精度是很难喝当前一些SOTA的LIO方法相比的，与其改这个，你不如直接修改其他的算法，比如FAST-LIO2.
> 
> 如果坚持要改LIO-SAM的话，其实github上已经有不少代码可以参考了，你看看这个 https://github.com/Gaochao-hit/LIO-SAM_based_relocalization

好的，感谢您的回答

kkkkkk812, 2025-05-06:
> [@kkkkkk812](https://github.com/kkkkkk812) 其实我不建议你这么做，LIO-SAM因为代码框架问题，目前只能做一些补丁。其鲁棒性和精度是很难喝当前一些SOTA的LIO方法相比的，与其改这个，你不如直接修改其他的算法，比如FAST-LIO2.
> 
> 如果坚持要改LIO-SAM的话，其实github上已经有不少代码可以参考了，你看看这个 https://github.com/Gaochao-hit/LIO-SAM_based_relocalization

你好，我现在还遇到了这样的问题，我想使用gps数据来进行辅助定位参考，但是会出现的问题是我在使用gps的时候，gps使用gtsam后，出现这样的情况，建图的时候，经纬度里程计和激光的里程计会有一个很大的方向偏差，大约90度，然后过了一会儿，激光里程计的方向会做一个很大角度的突然调整，然后后续就按照这个方向来走了，应该是激光的里程计的方向被RTK的强制纠正过来了，这是为什么呢

JokerJohn, 2025-05-06:
@kkkkkk812 这是因为，在某些场景中，起点一段GNSS可能效果不好，比如数据集中的hkust bag，如果这个时候加入了强gnss约束，很可能导致factor graph直接崩溃。 而且这是建图框架，所以在代码中，我把初始化流程设置得比较长，有足够的GNSS约束才会进行初始化（20个），进而对齐轨迹。

JokerJohn, 2025-05-06:
> [@JokerJohn](https://github.com/JokerJohn) 那我正在基于这套代码增加一个功能，将路侧的观测加入到因子图中，已经快完工了。这么说是不是基于fastlio2效果可能会更好？

@zhao-zhibo 看看我这个仓库 https://github.com/JokerJohn/MS-Mapping 的建图代码 （只使用single session mode），可以直接用，这也是我自用的，你再后端把GNSS约束加上就行。

JokerJohn, 2025-05-06:
> > [@kkkkkk812](https://github.com/kkkkkk812) 其实我不建议你这么做，LIO-SAM因为代码框架问题，目前只能做一些补丁。其鲁棒性和精度是很难喝当前一些SOTA的LIO方法相比的，与其改这个，你不如直接修改其他的算法，比如FAST-LIO2.
> > 如果坚持要改LIO-SAM的话，其实github上已经有不少代码可以参考了，你看看这个 https://github.com/Gaochao-hit/LIO-SAM_based_relocalization
> 
> 好的，感谢您的回答

@kkkkkk812 再看看我主页中PALoc的代码，把PGO换成滑动窗口就行，关闭闭环检测，然后地图的分辨率降低到0.5米。

closed 2025-05-06 by JokerJohn

kkkkkk812, 2025-05-08:
> [@kkkkkk812](https://github.com/kkkkkk812) 这是因为，在某些场景中，起点一段GNSS可能效果不好，比如数据集中的hkust bag，如果这个时候加入了强gnss约束，很可能导致factor graph直接崩溃。 而且这是建图框架，所以在代码中，我把初始化流程设置得比较长，有足够的GNSS约束才会进行初始化（20个），进而对齐轨迹。
感谢您的回答，但是我现在遇到了这样的一个问题，就是在启动以后小车的base_link在走了一段距离以后是正确的，但是同时gps的里程计的路线和小车存在较大的方向的差别，类似这个图所示

[image]，您能知道这是为什么吗，是gps转换的时候出现什么姿态的错误吗