JokerJohn/LIO_SAM_6AXIS #106: double  free or corruption

zhao-zhibo, 2025-04-28:
现在报错信息是双倍释放内存，从报错的栈信息看，是GTSAM报出来的，我不知道我的环境哪里出了问题。我也已经卸载并重新安装了GTSAM和Eigen，但是仍然不生效。报出来的信息是
```
double  free  or  corruption(out)     Thread 1 "lio_sam_6axis_m" received signal SIGABRT, Aborted.     __GI_raise (sig=sig@entry=6)  at ../sysdeps/unix/sysv/linux/raise.c:50       50   ../sysdeps/unix/sysv/linux/raise.c:  没有那个文件或目  
```
用gdb打印了栈信息，可是还是找不到问题。请问有什么思路建议吗？

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff67be859 in __GI_abort () at abort.c:79
#2  0x00007ffff6829266 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff6953298 "%s\n") at ../sysdeps/posix/libc_fatal.c:156
#3  0x00007ffff68312fc in malloc_printerr (str=str@entry=0x7ffff6955670 "double free or corruption (out)") at malloc.c:5347
#4  0x00007ffff6832fa0 in _int_free (av=0x7ffff6988b80 <main_arena>, p=0x55555575a930, have_lock=<optimized out>) at malloc.c:4314
#5  0x00007ffff723285d in gtsam::NoiseModelFactor::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
#6  0x00007ffff7233b3c in gtsam::(anonymous namespace)::_LinearizeOneFactor::operator()(tbb::blocked_range<unsigned long> const&) const () from /usr/local/lib/libgtsam.so.4
#7  0x00007ffff7233f1d in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, gtsam::(anonymous namespace)::_LinearizeOneFactor, tbb::auto_partitioner const>::execute() () from /usr/local/lib/libgtsam.so.4
#8  0x00007ffff6d06545 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#9  0x00007ffff6d0680f in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#10 0x00007ffff6d03b68 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#11 0x00007ffff7235c9f in gtsam::NonlinearFactorGraph::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
#12 0x00007ffff7207922 in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::FastVector<unsigned long> const&, boost::optional<gtsam::FastMap<unsigned long, int> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, bool) () from /usr/local/lib/libgtsam.so.4
#13 0x000055555561add1 in mapOptimization::saveKeyFramesAndFactor (this=0x7fffffffbc80) at /usr/include/c++/9/bits/stl_vector.h:94
#14 0x000055555561e0be in mapOptimization::laserCloudInfoHandler (this=0x7fffffffbc80, msgIn=...) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:380
#15 0x00005555555e0b82 in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&>::operator() (a0=..., this=<optimized out>) at /usr/include/boost/function/function_template.hpp:677
#16 boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const>) (function_obj_ptr=..., a0=...) at /usr/include/boost/function/function_template.hpp:158
#17 0x00005555555e1dce in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::operator() (a0=..., this=0x555555744288) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:474
#18 ros::SubscriptionCallbackHelperT<boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&, void>::call (this=0x555555744280, params=...) at /opt/ros/noetic/include/ros/subscription_callback_helper.h:144
#19 0x00007ffff755d199 in ros::SubscriptionQueue::call() () from /opt/ros/noetic/lib/libroscpp.so
#20 0x00007ffff750b162 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/noetic/lib/libroscpp.so
#21 0x00007ffff750c873 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/noetic/lib/libroscpp.so
#22 0x00007ffff756002f in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/noetic/lib/libroscpp.so
#23 0x00007ffff754824f in ros::spin() () from /opt/ros/noetic/lib/libroscpp.so
#24 0x00005555555ae869 in main (argc=<optimized out>, argv=<optimized out>) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:2462

```

JokerJohn, 2025-04-28:
@zhao-zhibo 看这行代码
`#24 0x00005555555ae869 in main (argc=<optimized out>, argv=<optimized out>) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:2462
`
把GPS cov打印出来看看有没有0值的。

zhao-zhibo, 2025-04-28:
@JokerJohn 我这个错误有点棘手，我看了栈信息是gtsam里面的报错，感觉是gtsam的版本和Eigen的版本的问题，不过我卸载重装了还是没有找到问题所在。2462行是ros::spin()，我没有使用gps，所以应该和gps的处理没有关系吧，所以想问问你有没有遇到类似的问题。

[image]

JokerJohn, 2025-04-28:
@zhao-zhibo 安装GTSAM时，cmake里面有一个指令是use. system eigen，你有没有设置这个。

zhao-zhibo, 2025-04-28:
@JokerJohn 这个操作我也已经试过了，我对gtsam编译用的命令是`cmake .. -DGTSAM_USE_SYSTEM_EIGEN=ON -DCMAKE_PREFIX_PATH=/usr/include/eigen3`。我的gtsam是4.0.0版本，Eigen是3.3.7版本

zhao-zhibo, 2025-04-30:
问题记录

环境：Ubuntu20.04,   Eigen 3.40  20.04系统对应命令可安装3.3.7，Pcl 1.13.,gtsam版本为gtsam-4.0.0-alpha2。
操作：在这个系统上试过很多方法，比如改变gtsam的版本升级为4.2.0版本，将Eigen降低版本为3.2.0版本，升级PCL版本为最新的版本。又譬如去参考了FASTLIVO2上的报错( https://github.com/hku-mars/FAST-LIVO/issues/173 )，可是并没有解决，卡了我四天左右。

解决方案：
重装了系统，控制变量法找环境依赖问题。
- 先在我的Ubuntu系统重装了Eigen 3.3.7( 操作命令：`sudo apt-get install libeigen3-dev  # 自动安装头文件到 /usr/include/eigen3` )，上面这个默认安装的是3.3.7版本。
- 然后安装了最新版本的pcl，pcl版本对应的哈希值为`5f608cfb5397fe848c7d61c4ae5f5b1ab760ba80`，可以先下载官网的版本，然后进行命令`git fetch origin  &&  git checkout 5f608cfb5397fe848c7d61c4ae5f5b1ab760ba80`。
- 最后是最关键的，我使用的仍然是`gtsam-4.0.0-alpha2`上一次系统的gtsam，然后安装完成到电脑之后，`double free or corruption `的错误竟然完美复现了，为本以为错误会消失。但是接下来为的操作和以往不一样，之前删除历史版本的gtsam没有删除完全，我将通过`sudo make install`安装的文件全部手动删除了，这个构建的库，不能通过`sudo make uninstall`进行删除，因此我手动进行了删除，删除命令如下:
```
   16  sudo rm -rf /usr/local/include/gtsam*
   17  sudo rm -rf /usr/local/include/CppUnitLite
   18  sudo rm -rf /usr/local/lib/libgtsam*
   19  sudo rm -rf /usr/local/lib/libmetis*
   20  sudo rm -rf /usr/local/lib/libCppUnitLite*
   21  sudo rm -rf /usr/local/lib/cmake/GTSAM*
   22  sudo rm -rf /usr/local/lib/cmake/GTSAMCMakeTools
   23  sudo rm -f /usr/local/bin/wrap
   24  sudo rm -rf /usr/local/share/gtsam
   25  sudo rm -rf /usr/local/share/sophus  # 如果确认属于 GTSAM 相关
   26  sudo ldconfig
```
**之后安装了最新版本的gtsam，4.2.0-ros版本，哈希值为`f64f094babc62a081a64c27c229816978c61b40f`，可以通过命令查看哈希值，` git log -1 --pretty=format:"%H"`。构建过程如下：**
```
   56  cmake ..
   57  make -j16
   58  sudo make install
```
最终不再报这个错误了。

JokerJohn, 2025-04-30:
@zhao-zhibo 看来这个版本gtsam-4.0.0-alpha2多少存在一点问题。

zhao-zhibo, 2025-04-30:
@JokerJohn 是的 ，gtsam-4.0.0-alpha2 这个版本我是在CSDN上面参考别人安装的