JokerJohn/LIO_SAM_6AXIS #111: Question regarding GPS correction and IMU pre-integration

Josgonmar, 2025-08-02:
Hey!
I’m integrating GPS pose corrections into LIO-SAM’s frontend so they feed into IMU preintegration, but when a GPS correction causes a significant jump in pose (the diff between LO and GPS causes a "jump" on the path), the IMU state drifts and the graph optimization eventually fails.

How do you mitigate the drift caused by GPS jumps when IMU integration continues across discontinuities?
If not, do you recommend resetting the preintegration front-end (and optionally the backend graph) at correction time?

I’d appreciate insights from your implementation experience or recommendations on patching in such logic.

Thanks in advance! 😄

JokerJohn, 2025-08-02:
Hi @Josgonmar !

In fact, I do not recommend doing this. Among the three sensors LIDAR, IMU, and GNSS, GNSS measurement stability is far inferior to LIDAR and IMU, even if you pre-fuse IMU and GNSS before fusing them with LIDAR. Regarding your approach, I have the following insights:

Introducing GNSS into preintegration has little significance, unless you have some raw GNSS measurements, such as pseudorange, carrier phase, etc., and you're attempting to achieve tighter coupling. Additionally, the preintegration implementation in GTSAM is very unstable, and I recommend implementing it yourself.
GNSS exhibits path jumping when losing lock, especially in the z-direction. Without more raw observations, I recommend strictly limiting the introduction of GNSS information. Generally, I would first roughly understand GNSS observation quality through curve plots, then select a strict threshold to limit GNSS introduction, avoiding single outliers from causing the entire graph to collapse. To be safe, you can use a robust kernel when introducing GNSS factors to avoid extreme outliers.
I do not recommend introducing GNSS in the preintegration frontend. The main purpose of the frontend is to maintain stability and provide stable relative pose constraints.

This appears to be technical advice about sensor fusion in SLAM (Simultaneous Localization and Mapping) systems, specifically discussing the integration of LIDAR, IMU, and GNSS sensors with considerations about measurement stability and robust optimization techniques.

JokerJohn, 2025-08-02:
@Josgonmar 

Your observation is very detailed, and this problem indeed exists. This is caused by the coupling manner in LIO-SAM, which contains two factor graphs, where the factor graph of the preintegration part uses the optimized LiDAR poses from PGO to correct IMU states, and the new IMU states (bias, etc.) will be utilized to estimate the initial poses for subsequent matching. In the implementation of the preintegration factor graph, it can be found that to avoid computation being too slow, when LiDAR pose nodes exceed 500, the entire graph will be reset, resetting the IMU states. Only the marginalized results will be retained as priors for the new graph (as the first frame prior), which is different from standard sliding windows and is not an elegant approach. 
You can use the [FixedLagSmoother provided in GTSAM ](https://github.com/JokerJohn/gtsam/blob/develop/gtsam_unstable/examples/FixedLagSmootherExample.cpp) to achieve this purpose. However, although this standard sliding window only retains the latest states, it still contains partial information from the marginalized old states. This partial information contains accumulated errors, and as the motion distance increases, this accumulated error becomes more apparent. Therefore, periodically resetting the graph is actually a simpler approach, but it inevitably brings optimization instability.
For this point, there is another tightly coupling manner for LiDAR and IMU, similar to **LIO-Mapping**, which jointly optimizes all IMU and LiDAR residuals together, rather than optimizing LiDAR and IMU in separate steps like the current repo, where IMU is only used to predict states and does not enter the optimization residuals of LiDAR matching.
This issue is currently controversial. I disagree with LIO-Mapping's approach because I believe that introduce of IMU is generally effective for high-frequency intense motion, mainly due to providing more accurate initial poses, while joint optimization may actually compromise LiDAR matching accuracy and be extremely slow in optimization. 
**FAST-LIO2** also adopts this step-by-step approach: IMU prediction, LiDAR update. This tells us that preintegration is not essential for LIO; ordinary integration can still achieve high robustness and precision.

Josgonmar, 2025-08-02:
@JokerJohn 

Thanks a lot for your insights! I'll explore the options and determine the best approach for my use case. Appreciate your help 😄